morrownr/8821cu-20210916 #161: unable to compile cause of error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’?

creatxrgithub, 2024-10-02:
6.11.1-2-cachyos

```
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2031:17: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Wimplicit-function-declaration]
 2031 |                 strlcpy(info->driver, wiphy_dev(wdev->wiphy)->driver->name,
      |                 ^~~~~~~
      |                 strncpy
```

it has some other warnings. only one error.

creatxrgithub, 2024-10-02:
full console message

```
❯ make
/bin/sh: line 1: bc: command not found
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.11.1-2-cachyos/build M=DRIVER/8821cu-20210916  modules
make[1]: Entering directory '/usr/lib/modules/6.11.1-2-cachyos/build'
/bin/sh: line 1: bc: command not found
  CC [M]  DRIVER/8821cu-20210916/os_dep/linux/os_intfs.o
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1087:6: warning: no previous prototype for ‘rtw_regsty_load_target_tx_power’ [-Wmissing-prototypes]
 1087 | void rtw_regsty_load_target_tx_power(struct registry_priv *regsty)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1681:14: warning: no previous prototype for ‘rtw_classify8021d’ [-Wmissing-prototypes]
 1681 | unsigned int rtw_classify8021d(struct sk_buff *skb)
      |              ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1817:5: warning: no previous prototype for ‘rtw_ndev_init’ [-Wmissing-prototypes]
 1817 | int rtw_ndev_init(struct net_device *dev)
      |     ^~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1830:6: warning: no previous prototype for ‘rtw_ndev_uninit’ [-Wmissing-prototypes]
 1830 | void rtw_ndev_uninit(struct net_device *dev)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1893:6: warning: no previous prototype for ‘rtw_hook_if_ops’ [-Wmissing-prototypes]
 1893 | void rtw_hook_if_ops(struct net_device *ndev)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1975:5: warning: no previous prototype for ‘rtw_os_ndev_alloc’ [-Wmissing-prototypes]
 1975 | int rtw_os_ndev_alloc(_adapter *adapter)
      |     ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c: In function ‘rtw_ethtool_get_drvinfo’:
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2031:17: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Wimplicit-function-declaration]
 2031 |                 strlcpy(info->driver, wiphy_dev(wdev->wiphy)->driver->name,
      |                 ^~~~~~~
      |                 strncpy
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c: At top level:
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2130:5: warning: no previous prototype for ‘rtw_os_ndev_register’ [-Wmissing-prototypes]
 2130 | int rtw_os_ndev_register(_adapter *adapter, const char *name)
      |     ^~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2281:5: warning: no previous prototype for ‘rtw_os_ndevs_alloc’ [-Wmissing-prototypes]
 2281 | int rtw_os_ndevs_alloc(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2333:6: warning: no previous prototype for ‘rtw_os_ndevs_free’ [-Wmissing-prototypes]
 2333 | void rtw_os_ndevs_free(struct dvobj_priv *dvobj)
      |      ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2465:4: warning: no previous prototype for ‘rtw_init_default_value’ [-Wmissing-prototypes]
 2465 | u8 rtw_init_default_value(_adapter *padapter)
      |    ^~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:3705:5: warning: no previous prototype for ‘rtw_os_ndevs_register’ [-Wmissing-prototypes]
 3705 | int rtw_os_ndevs_register(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:3822:6: warning: no previous prototype for ‘netdev_br_init’ [-Wmissing-prototypes]
 3822 | void netdev_br_init(struct net_device *netdev)
      |      ^~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4139:6: warning: no previous prototype for ‘ips_netdrv_open’ [-Wmissing-prototypes]
 4139 | int  ips_netdrv_open(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4332:5: warning: no previous prototype for ‘pm_netdev_open’ [-Wmissing-prototypes]
 4332 | int pm_netdev_open(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4460:5: warning: no previous prototype for ‘pm_netdev_close’ [-Wmissing-prototypes]
 4460 | int pm_netdev_close(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5158:5: warning: no previous prototype for ‘rtw_suspend_normal’ [-Wmissing-prototypes]
 5158 | int rtw_suspend_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5571:6: warning: no previous prototype for ‘rtw_mi_resume_process_normal’ [-Wmissing-prototypes]
 5571 | void rtw_mi_resume_process_normal(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5604:5: warning: no previous prototype for ‘rtw_resume_process_normal’ [-Wmissing-prototypes]
 5604 | int rtw_resume_process_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: DRIVER/8821cu-20210916/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/lib/modules/6.11.1-2-cachyos/build/Makefile:1934: DRIVER/8821cu-20210916] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.1-2-cachyos/build'
make: *** [Makefile:2501: modules] Error 2
```

morrownr, 2024-10-02:
Hi @creatxrgithub 

> /bin/sh: line 1: bc: command not found

You need to install the `bc` package.

> /lib/modules/6.11.1-2-cachyos/build

I am not familiar with `cachyos`. Is it Debian based? More details would help.

What kernal version and gcc version is in use?

creatxrgithub, 2024-10-03:
cachyos is arch linux based distro
kernel version: 6.11.1-2
gcc version 14.2.1 20240910 (GCC) 

after installed bc, the full console message:
```
❯ make
make ARCH=x86_64 CROSS_COMPILE= -C /lib/modules/6.11.1-2-cachyos/build M=DRIVER/8821cu-20210916  modules
make[1]: Entering directory '/usr/lib/modules/6.11.1-2-cachyos/build'
  CC [M]  DRIVER/8821cu-20210916/os_dep/linux/os_intfs.o
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1087:6: warning: no previous prototype for ‘rtw_regsty_load_target_tx_power’ [-Wmissing-prototypes]
 1087 | void rtw_regsty_load_target_tx_power(struct registry_priv *regsty)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1681:14: warning: no previous prototype for ‘rtw_classify8021d’ [-Wmissing-prototypes]
 1681 | unsigned int rtw_classify8021d(struct sk_buff *skb)
      |              ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1817:5: warning: no previous prototype for ‘rtw_ndev_init’ [-Wmissing-prototypes]
 1817 | int rtw_ndev_init(struct net_device *dev)
      |     ^~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1830:6: warning: no previous prototype for ‘rtw_ndev_uninit’ [-Wmissing-prototypes]
 1830 | void rtw_ndev_uninit(struct net_device *dev)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1893:6: warning: no previous prototype for ‘rtw_hook_if_ops’ [-Wmissing-prototypes]
 1893 | void rtw_hook_if_ops(struct net_device *ndev)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:1975:5: warning: no previous prototype for ‘rtw_os_ndev_alloc’ [-Wmissing-prototypes]
 1975 | int rtw_os_ndev_alloc(_adapter *adapter)
      |     ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c: In function ‘rtw_ethtool_get_drvinfo’:
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2031:17: error: implicit declaration of function ‘strlcpy’; did you mean ‘strncpy’? [-Wimplicit-function-declaration]
 2031 |                 strlcpy(info->driver, wiphy_dev(wdev->wiphy)->driver->name,
      |                 ^~~~~~~
      |                 strncpy
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c: At top level:
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2130:5: warning: no previous prototype for ‘rtw_os_ndev_register’ [-Wmissing-prototypes]
 2130 | int rtw_os_ndev_register(_adapter *adapter, const char *name)
      |     ^~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2281:5: warning: no previous prototype for ‘rtw_os_ndevs_alloc’ [-Wmissing-prototypes]
 2281 | int rtw_os_ndevs_alloc(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2333:6: warning: no previous prototype for ‘rtw_os_ndevs_free’ [-Wmissing-prototypes]
 2333 | void rtw_os_ndevs_free(struct dvobj_priv *dvobj)
      |      ^~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:2465:4: warning: no previous prototype for ‘rtw_init_default_value’ [-Wmissing-prototypes]
 2465 | u8 rtw_init_default_value(_adapter *padapter)
      |    ^~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:3705:5: warning: no previous prototype for ‘rtw_os_ndevs_register’ [-Wmissing-prototypes]
 3705 | int rtw_os_ndevs_register(struct dvobj_priv *dvobj)
      |     ^~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:3822:6: warning: no previous prototype for ‘netdev_br_init’ [-Wmissing-prototypes]
 3822 | void netdev_br_init(struct net_device *netdev)
      |      ^~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4139:6: warning: no previous prototype for ‘ips_netdrv_open’ [-Wmissing-prototypes]
 4139 | int  ips_netdrv_open(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4332:5: warning: no previous prototype for ‘pm_netdev_open’ [-Wmissing-prototypes]
 4332 | int pm_netdev_open(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:4460:5: warning: no previous prototype for ‘pm_netdev_close’ [-Wmissing-prototypes]
 4460 | int pm_netdev_close(struct net_device *pnetdev, u8 bnormal)
      |     ^~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5158:5: warning: no previous prototype for ‘rtw_suspend_normal’ [-Wmissing-prototypes]
 5158 | int rtw_suspend_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5571:6: warning: no previous prototype for ‘rtw_mi_resume_process_normal’ [-Wmissing-prototypes]
 5571 | void rtw_mi_resume_process_normal(_adapter *padapter)
      |      ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
DRIVER/8821cu-20210916/os_dep/linux/os_intfs.c:5604:5: warning: no previous prototype for ‘rtw_resume_process_normal’ [-Wmissing-prototypes]
 5604 | int rtw_resume_process_normal(_adapter *padapter)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [scripts/Makefile.build:244: DRIVER/8821cu-20210916/os_dep/linux/os_intfs.o] Error 1
make[2]: *** [/usr/lib/modules/6.11.1-2-cachyos/build/Makefile:1934: DRIVER/8821cu-20210916] Error 2
make[1]: *** [Makefile:224: __sub-make] Error 2
make[1]: Leaving directory '/usr/lib/modules/6.11.1-2-cachyos/build'
make: *** [Makefile:2501: modules] Error 2
```

morrownr, 2024-10-03:
> kernel version: 6.11.1-2, gcc version 14.2.1 20240910 (GCC)

Okay, you are on the bleeding edge. I need to work on a test system to get it up to kernel 6.11 and gcc 14. I don't think this will take long once I am ready to start but first...

I am curious why you are not using the in-kernel rtw88 rtw88_8821cu driver? I've been working on a small team that has been making big improvements to the rtw88 driver since late last year, Did you try the adapter before installing this driver?

To try it, simply uninstall this driver, which will blacklist the rtw88_8821cu driver so that the don't conflict. I could use a report so I would appreciate it if you test rtw88_8821cu.

creatxrgithub, 2024-10-03:
i use a debian/ubuntu based distros before. it has not driver at that time. so, i found your project. thanks.

and i installed cachyos several days ago. it doesn't auto install driver for the rtl 8821cu usb wifi + bluetooth card. (Chipset: Realtek RTL8811AU.)

so, i think it may not have driver in the distro until now. i didn't find it in repo.

morrownr, 2024-10-03:
> so, i think it may not have driver in the distro until now. i didn't find it in repo.

Kernel 6.11 has the driver unless it is specifically not being compiled, which I doubt. We would also need to check to make sure the firmware files are in place. The quickest way to see if rtw88_8821au is installed is to first remove the driver in this repo and reboot with the adapter in a port.:

$ sudo sh remove-driver.sh
$ sudo reboot

$ sudo lsmod | grep rtw88

Post the result.

creatxrgithub, 2024-10-03:
i asked the company before. the chipset is  Realtek RTL8811AU.

this driver (https://github.com/lwfinger/rtw88), i tried again, it's not working for me. it has not 8811au driver

in fact, i've tried several times before, i found only yours works.
```
❯ sudo lsmod | grep rtw88
[sudo] password for creatxr: 
rtw88_8821cu           12288  0
rtw88_8821c            98304  1 rtw88_8821cu
rtw88_usb              32768  1 rtw88_8821cu
rtw88_core            348160  2 rtw88_8821c,rtw88_usb
mac80211             1724416  2 rtw88_core,rtw88_usb
cfg80211             1421312  2 rtw88_core,mac80211
```

morrownr, 2024-10-03:
> rtw88_8821cu           12288  0

There is the rtw88 driver for rtl8821/11cu chips.

I am seeing some inconsistency in the thread here. We need to verify what chip your adapter uses. Run and post the results of:

$ lsusb

creatxrgithub, 2024-10-03:
```
Bus 001 Device 002: ID 0bda:c820 Realtek Semiconductor Corp. 802.11ac NIC
```

creatxrgithub, 2024-10-03:
oh. sorry. i think i made a very big mistake. the wireless card which does not installed driver is my internal bcm43142 .

you are right. i am using in-kernel driver right now.

sorry.

creatxrgithub, 2024-10-03:
> See if it does what you need it to do. Using the in-kernel drivers is preferable to out-of-kernel drivers. Let me know.
yes, it works. i unplug the usb adapter just now. i found my mistake. sorry. 
i'll check if bluetooth works.
now, i installed broadcom-wl-dkms and the two wireless card are shown in the network-manager.

> FYI: A few minutes ago I compiled the driver in this repo on Ubuntu 24.04 with kernel 6.11 and gcc 14. I was expecting to see what you posted but it was a clean compile. That leaves me stumped.

cachyos or arch linux may has something different. i can compile this driver with ubuntu based bodhi linux and it works well.

creatxrgithub, 2024-10-03:
after installed bluez and bluez-utils and blueman, then "systemctl start bluetooth", bluetooth is working now.
cause the bluetooth didn't installed and start by default, i mistakenly thought it was that internal bcm43142 in working.

morrownr, 2024-10-03:
Great! I think you learned a few things today and I hope you continue to have success. The in-kernel rtw88 driver for your adapter should be in good shape as long as you are using a 6.11 or later kernel. 6.12 will have additional improvements. As long as rtw88_8821cu works for you, it is the in-kernel maintained driver so you should not use this out-of-kernel driver. Glad to see BT is working as well.

I do think  cachyos or arch linux has something different that is causing the compile problem but I need to move on to help others and will leave the Arch Linux troubleshooting to others.

Cheers

dubhater, 2024-10-03:
strlcpy was removed: https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/include/linux/string.h?h=v6.11.1&id=d26270061ae66b915138af7cd73ca6f8b85e6b44

morrownr, 2024-10-03:
@dubhater 

Thanks for the info but I was aware of this change with strlcpy and have a fix worked up... however, since he was using kernel 6.11 I wanted to see if he would be satisfied with rtww88_8821cu. I guess we will see. I have sort of lost track of when the various patches have gone in but it seems like rtw88_8821cu should be in good enough shape for most use cases. What are your thoughts?

It is interesting that strlcpy errored on Arch but not on Ubuntu with the same level of kernel and compiler.

dubhater, 2024-10-04:
It should work. Maybe slower than normal.