morrownr/8821cu-20210916 #162: Poor performance while connected to a 2,4 GHz wifi network and a bluetooth device

PnY99, 2024-10-05:
Hello,
I have a USB WiFi/Bluetooth adapter manifactured by TP-Link with the rtl8821cu chipset and I'm encountering this problem when using your drivers.
If I'm connected to a 5 GHz network the performance is fine, regardless of the bluetooth state.
If I switch to a 2,4 GHz network, performances are good only if there are no bluetooth devices connected to the adapter. If I connect a bluetooth mouse, the WiFi rate immediately drops at 6 Mbit/s and the mouse behaviour is laggy.
I understand that the bluetooth shares the same radio/antenna with the 2,4 GHz WiFi, but is there anything that can be done to solve the issue? I can post more information if needed
Thank you

morrownr, 2024-10-05:
Hi @PnY99 

> but is there anything that can be done to solve the issue?

This is one of the reasons that I recommend users only get wifi adapters that are wifi only. I get that BT is desirable but my experience is that it is best to get a separate nano BT only adapter that can be placed in another port that is not close to the wifi adapter.

The Main Menu for this site is:

https://github.com/morrownr/USB-WiFi

Read menu item 1 as I mention that considering the purchase of wifi adapters that also support BT is not a good idea. There is also a big problem with USB3 capable wifi adapters and BT as USB3 cables and connections emit a signal that interferes with BT.

Your best option is to use 5 GHz for wifi as there should be no interference.

PnY99, 2024-10-07:
Unfortunately I can only allocate a single USB port to the WiFi/Bluetooth adapter so purchasing a separate device is not an option. I will try not to use a 2,4 GHz WiFi network when using a bluetooth device, if possible. Thank you for the reply

closed 2024-10-07 by PnY99

dubhater, 2024-10-07:
How does it perform with rtw88?

morrownr, 2024-10-07:
@PnY99 

> Unfortunately I can only allocate a single USB port to the WiFi/Bluetooth adapter...

While i prefer USB adapters for wifi, you may have a situation where a card could be your best choice. In my main dev box here, I use a mt7922 based PCIe card. It smokes at over 1 Gbps with wifi 6 and 160 MHz channel width. This is a desktop system. The card has bt capability that just works wonderfully. There is also the option of a M.2 or other card format if your system is a laptop. I have a mt7921 M.2 card in my laptop and it works wonderfully, wifi and bt. I have a mini pc that has a mt7925 based M.2 card and it works wonderfully. The mt7925 is a WiFi 7 capable chip. All of these cards I am talking about are Plug and Play in modern Linux distros. There is no need to find or install a driver.

You might want to see what happens with the in-kernel rtw88 driver for your adapter as @dubhater mentioned. It might be better or it might not. I've been watching this problem for years and it is not specific to Linux. It mostly is caused by engineering issues with the two different technologies and also sometimes with poor implementation by adapter makers. You don't see the problem with mixing wifi and bt on cards because the wifi on cards is going through the fast bus, not the slower usb bus.

Hopefully something in this message helps.