morrownr/8821cu-20210916 #163: Driver for RTL8812CU (0bda:c812)

tarotten10, 2024-10-07:
Hello. I realize I'm writing about another device, but is it possible for you to add support?

morrownr, 2024-10-07:
Hi @tarotten10 

I have been working with a small team of devs and testers this year to improve RTW88. There is a driver for your device in RTW88 in the kernel. My recommendation is that you use a distro with a kernel that is as modern as possible and your adapter should be plug and play.

A good distro as an example would be the soon to be released Ubuntu 24.10 that uses kernel 6.11... or any other distro that uses kernel 6.11. Let me know how that works out.

tarotten10, 2024-10-07:
Your work is really awesome. Is there anything I can do to help if I'm not a developer? 
I'm using Fedora and it doesn't work there :(
Any ideas? And what about Arch, Kali?

morrownr, 2024-10-07:
What kernel are you using with Fedora?

$ uname -r

Also, provide the output of the follow command so we can check if your vid-pid is in the kernel

$ lsusb

> Is there anything I can do to help if I'm not a developer?

Yes, once we get you going, test it and report.

tarotten10, 2024-10-08:
[image]

morrownr, 2024-10-08:
I think I have an idea what we need to do to get your adapter going and the problem in the kernel is actually my fault. I have send a patch in to fix the issue and it is going into kernel 6.12 and will be backported to actively supported kernels but 6.10 is no longer supported I think so it may not work its way into 6.10. We can come up with a workaround.

Show me the results of the following:

$ lsmod | grep rtw

I think we need to setup a blacklist until you are on kernel 6.12 or later.

morrownr, 2024-10-09:
> 	{ USB_DEVICE_AND_INTERFACE_INFO(RTW_USB_VENDOR_ID_REALTEK, 0xc812, 0xff, 0xff, 0xff),

I just checked in the Linux kernel and the vid-pid for your adapter is there so rtw88_8822cu (rtw88) should be driving your adapter... unless you have blacklisted it. Have you installed any other drivers? Or manually set up a blacklist?

tarotten10, 2024-10-11:
I only installed a driver from this repository and then tried another one for RTL8811 devices, but uninstalled them. About the blacklist - no, I didn't put anything in there.
I'm not too sure either way.