morrownr/8821cu-20210916 #164: Try to connect, it  does not ask for a password and  say "connection lost"

62Maurizio, 2024-10-25:
Env: Pc  dual booting win7sp1-x64  and debian 12.7 kernel 6.1.0-25amd64  lxqt desktop manager
try to connect device Relatek ven 0bda c811  to android12 smartphone wifi hotspot  
Win7 is able to connect correctly with the smartphone 
i follow your procedure: ,i installed all prereq with synaptic , i downloaded your files with git command and then i succesfully run install script, the drive installs without errors, then as instructoed i reboot , the usb wifi blinks  and the  nm-tray correctly see all wifi networks,  
but when i try connect to  my smartphone, nm-tray does not ask for a password, but shows a connection lost pop-up 
if i try with  wifi networks of my  neighborhoods  nm-tray ask me correctly for a password... (i tried every network)
a this point i checked my cell, it permits up to 10 connection ,and uses wpa2/wpa3 protection , i reboot also the cell but the situation  is always the same, i uninstall and then install again the driver but situation does not change
what is worng between my cell and the driver ?
thanks in advance for your help
Maurizio

62Maurizio, 2024-10-25:
i load here the requested commnad output 
[dkms.txt](https://github.com/user-attachments/files/17523556/dkms.txt)
[iw.txt](https://github.com/user-attachments/files/17523558/iw.txt)
[iwreg.txt](https://github.com/user-attachments/files/17523559/iwreg.txt)
[lsusb.txt](https://github.com/user-attachments/files/17523560/lsusb.txt)
[rfkill.txt](https://github.com/user-attachments/files/17523561/rfkill.txt)

62Maurizio, 2024-10-25:
Ok loaded... it seems all correct

closed 2024-10-25 by 62Maurizio

morrownr, 2024-10-25:
Hi @62Maurizio 

> uses wpa2/wpa3 protection

I've seen a lot of problems with wpa2/wpa3 mixed set as security. It isn't just a Linux problem. If you have any clients that would connect to your phone that are not wpa3 capable, then you best option may be to use wpa2. If everything does support wpa3, then set wpa3.

FWIW: On Debian once, once Debian 13 is out with kernel 6.12, you should not longer need this driver as the rtw88 in-kernel driver is really good starting with kernel 6.12... plug and play.

62Maurizio, 2024-10-26:
Hi  
I set to wpa2, and it still no prompt me for the password  and i get the connection lost pop-up
then i unprotect the connection and i still get the connection lost 
then i look to nm-tray menu and it shows the connection to my cell as a knowed connection and it is signed with a "X"
i suspected that somewere this connection is flagged as broken... so i uninstalled nm-tray and i reinstalled again but the results are always the same, so IMHO the problem is inside debian that flags somewere the connection as  broken... there is no other possibility because i tried all the connection of my neighbors and it ALWAYS prompt me for the password ....
Before proceeding to a new fresh installation have you any idea about where the connection is  flagged ?
Thanks in adavence for the support 
Maurizio

62Maurizio, 2024-10-26:
Ok I fixed ! 
I completely remove nm-tray program and libraries , and i install gnome connection manager , now i'm able to connect,.
The problem was inside nm-tray that somewere flagged the connection ..

morrownr, 2024-10-26:
Ah. Okay, have you reported this to your distro?