morrownr/8821cu-20210916 #169: 0bda:c811 works on Raspberry Pi 2

potassium-chloride, 2024-12-14:
Host device: Raspberry Pi 2 Model B Rev 1.1
OS: Raspbian GNU/Linux 12 (bookworm) Lite
Kernel: 6.6.62+rpt-rpi-v7

WiFi device:
`0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)` (first plug)
`0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC` (after modeswitch or reconnection)
MAC identification: 90-DE-80, Shenzhen Century Xinyang Technology Co.
[image]

Connection has some troubles. `sudo raspi-config` cannot connect. I connect throw:
`sudo modprobe ...` (some non 8821cu modules: mac80211, rtw88, rtw88_core, rtw88_usb, probably it wasn't necessary?..)
`sudo ifconfig wlan0 up`,
`sudo wpa_cli ...`,
`sudo dhclient -r`,
`sudo dhclient wlan0`.
IPv6 address doesn't work:
`Dec 14 21:36:08 raspberrypi dhclient[4969]: no link-local IPv6 address for wlan0` (How can I fix it?)

And know it works! Link Quality=73/100  Signal level=39/100  Noise level=0/100
Unfortunately, It doesn't work pretty well:
For a few first minutes there was a lot of loses (15-50% loses while pings). After that link was more stable but still not good. Speedtest showed 11-92 Mbit/s (maximum is 100 Mbit/s in my network). My phone in same conditions has got 91-96 Mbit/s.

morrownr, 2024-12-15:
Hi @potassium-chloride 

> rtw88, rtw88_core, rtw88_usb

Yes, driver support for this chip was added starting with kernel 6.2. My driver should automatically blacklist the in-kernel driver if you installed with the install-driver.sh script. Let me go take a look...

The in-kernel driver was originally not that good but a lot of work over the last year has it in good shape now in kernel 6.12 and later. Hopefully RasPiOS is upgraded to kernel 6.12 in the next few months.

potassium-chloride, 2024-12-15:
I have tried connecting the WiFi adapter before I found this repository and it was not detected by the system in any way, no kernel modules were loaded while connected. Although I found a rtl8821cu.ko file in the Raspbian image, but it was not used by Linux in any my experiments.

By the way, when I plugged WiFi adapter at second time interface turned up automatically. It is better but still not useful for automatic systems.

morrownr, 2024-12-15:
> rtl8821cu.ko

I don't recognize that driver name. The default driver name is 8821cu.ko for these vendor drivers and rtw88_8821cu.ko for the in-kernel driver. Maybe there is a problem with a driver that has been installed but not uninstalled.

A little history: I've seen more bug reports on this rtl8821/11cu chip than any other chip I've tried to support. While work on the in-kernel driver over this last year, it appears that some adapter makers burned the eeprom with bad info when making the adapters. Then there is the issue of the adapter being multi-state. Here are a couple of ideas:

- uninstall this driver by running sudo sh remove-driver.sh

- install the downstream rtw88 driver from the following repo:

https://github.com/lwfinger/rtw88

- take a look at the Main Menu of this site. It contains a lot of info. Menu items one and two are good reading:

https://github.com/morrownr/USB-WiFi