morrownr/8821cu-20210916 #170: error: ‘struct usb_driver’ has no member named ‘driver’

sharatatwork, 2024-12-16:
Debian 12 fresh install
6.1.0-27-amd64 
Unable to compile
`
error: ‘struct usb_driver’ has no member named ‘driver’
`
`
/var/lib/dkms/rtl8821cu/5.12.0.4/build/os_dep/linux/usb_intf.c:342:17: error: ‘struct usb_driver’ has no member named ‘driver’
  342 |         .usbdrv.driver.shutdown = rtw_dev_shutdown,
      |                 ^~~~~~
/var/lib/dkms/rtl8821cu/5.12.0.4/build/os_dep/linux/usb_intf.c:342:35: error: initialization of ‘const char *’ from incompatible pointer type ‘void (*)(struct device *)’ [-Werror=incompatible-pointer-types]
  342 |         .usbdrv.driver.shutdown = rtw_dev_shutdown,

  CC [M]  /var/lib/dkms/rtl8821cu/5.12.0.4/build/os_dep/linux/recv_linux.o
make[2]: ***
`

morrownr, 2024-12-16:
Hi @sharatatwork 

Did you follow the Installations Steps in the README?

Are you installing with `sudo sh install-driver.sh` ?

I suspect you did not install all requirements for the dev environment.

Sauerlandlinux, 2024-12-16:
It's a regression of the last commit.

I think, the last commit was made for a backported Rocky Linux Kernel.

morrownr, 2024-12-16:
> It's a regression of the last commit.

Got it. I have now reverted that PR. Let me know if that fixes the problem.

Sauerlandlinux, 2024-12-16:
Thank you for reverting.

I am building the kmps for openSUSE now for years and the last commit has also broken at the same place as in sharatatwork's post.

So I looked inside the last patch and there are some pieces, I use for patching Leap (15.5, 15.6 and 16.0).
Thinking about a complete revert of the patch in my builds, but now, they will build fine as before.
All ok on openSUSE.

Waiting what sharatatwork will answer.

PS:
Thank you for helping in development of the drivers after Larry Fingers passing.

morrownr, 2024-12-16:
Okay, thanks for the good news. 

> Thank you for helping in development of the drivers after Larry Fingers passing.

Hated to see Larry go but you just have to press on. I am hoping to stop updating these WiFi 5 drivers once all of the work is in the kernel as the new in-kernel drivers should prove to be better than these Realtek drivers, The only remaining WiFi 5 driver is the 8814au and it is working but not ready to push into the downstream repo yet. That should just about close out future need for these WiFi 5 out-of-kernel drivers.

USB WiFi has come a long way in the last few years.

Sauerlandlinux, 2024-12-16:
Many has changed in the last few years.
Many drivers are now in the Kernel such as rtw88 and rtw89 (with the newer ones, thank you).

But as long, as Distributions use an "older" kernel, I will build them.......
Leap 16.0 has also a backported kernel 6.4.........

morrownr, 2024-12-16:
> as Distributions use an "older" kernel, I will build them.......

That is why I plan to leave the drivers up but simply discontinue API updates with kernel 6.12. It will take time for the new drivers to be fully utilized. If you have any suggestions, let me know.

sharatatwork, 2024-12-18:
Sorry for the delay.  Got  sidetracked.
But yes, compilation went through.  many thanks @Sauerlandlinux  @morrownr 
(I have not rebooted yet, will do. Just a insmod did not work, probably that's expected.)

sharatatwork, 2024-12-18:
Rebooted. The device works as expected. I hope I can close this.

closed 2024-12-18 by sharatatwork

morrownr, 2024-12-18:
@sharatatwork 

Glad it is working.

@Sauerlandlinux 

You might want to nudge the person that handles firmware for OpenSuse to start including the firmware for rtl8812au and rtl8821/11au s the new drivers just went into 6.13.

https://git.kernel.org/pub/scm/linux/kernel/git/firmware/linux-firmware.git/tree/rtw88

Sauerlandlinux, 2024-12-18:
You get one or two times a month new firmware in Tumbleweed, for Leap firmware is offered in some Repos you have to add by yourself.
The last Firmware is from 11 December 2024 from linux-firmware git tree.