morrownr/8821cu-20210916 #171: PC "freeze" sometimes when a sr0 CD-ROM is detected

Massedil, 2024-12-16:
Hello,

Since I installed this driver, when I boot with the USB Wi-Fi adapter connected, sometimes (once out of 2 boot ?) my computer "freeze" at the login screen. It is not really a freeze because the displayed hour is updated, but I can't use my keyboard and my mouse. If I press the power button, the computer try to stops, but hangs.

Comparing the log between a succesfull boot (I can login and use my computer) and a failed one (stuck at boot screen), I can see that the kernel dectect a `sr0` `CD-ROM            Realtek  Driver Storage` when it fails.

Here are the logs : 

```
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 16 15:38:27 XXXXXXXXXX kernel: scsi 8:0:0:0: CD-ROM            Realtek  Driver Storage   1.00 PQ: 0 ANSI: 0 CCS
déc. 16 15:38:27 XXXXXXXXXX kernel: sr 8:0:0:0: [sr0] scsi3-mmc drive: 0x/0x caddy
déc. 16 15:38:27 XXXXXXXXXX kernel: cdrom: Uniform CD-ROM driver Revision: 3.20
déc. 16 15:38:27 XXXXXXXXXX kernel: sr 8:0:0:0: Attached scsi CD-ROM sr0
```

When I try to stop the computer, it fails to stop and I see this : 

```
déc. 16 15:39:27 XXXXXXXXXX systemd-udevd[379]: sr0: Worker [483] processing SEQNUM=3231 is taking a long time
déc. 16 15:41:27 XXXXXXXXXX (udev-worker)[483]: sr0: Spawned process 'cdrom_id --lock-media /dev/sr0' [1642] timed out after 46s, killing
déc. 16 15:41:27 XXXXXXXXXX systemd-udevd[379]: sr0: Worker [483] processing SEQNUM=3231 killed
déc. 16 15:41:27 XXXXXXXXXX systemd-udevd[379]: sr0: Worker [483] terminated by signal 9 (KILL).
déc. 16 15:42:28 XXXXXXXXXX systemd-udevd[379]: sr0: Worker [1643] processing SEQNUM=3235 is taking a long time
```

- Do you know how I can solve this ?

---
Ubuntu 24.04.1 LTS / kernel 6.8.0-50-generic / USB : `0bda:c811 Realtek Semiconductor Corp. 802.11ac NIC`

--- 

More logs : 

boot failed : 

```
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: new high-speed USB device number 2 using xhci_hcd
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: Product: DISK
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:38:58 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:39:28 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:39:59 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:40:30 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:40:40 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:41:11 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:41:41 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:42:12 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:42:43 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:43:14 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:43:44 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:44:15 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:44:22 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
```

Boot successful : 

```
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: new high-speed USB device number 2 using xhci_hcd
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: Product: DISK
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 16 15:47:01 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:48:40 XXXXXXXXXX kernel: usb 1-5: USB disconnect, device number 2
déc. 16 15:48:43 XXXXXXXXXX kernel: usb 1-5: new high-speed USB device number 6 using xhci_hcd
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: Product: DISK
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: USB disconnect, device number 6
déc. 16 15:48:44 XXXXXXXXXX kernel: usb 1-5: new high-speed USB device number 7 using xhci_hcd
déc. 16 15:48:45 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
déc. 16 15:48:45 XXXXXXXXXX kernel: usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=3
déc. 16 15:48:45 XXXXXXXXXX kernel: usb 1-5: Product: 802.11ac NIC
déc. 16 15:48:45 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 16 15:48:45 XXXXXXXXXX kernel: usb 1-5: SerialNumber: 123456
```

I don't know if it is related, but even when the boot is successful, I often have to disconnect and reconnect the USB device for it to connect to Wi-Fi networks.

I'm available for tests if you need !

morrownr, 2024-12-16:
Hi @Massedil 

> Do you know how I can solve this ?

I'll try to help. This is actually not a driver issue. It is a problem with the adapter being multi-state.

I have recently been looking into this issue as there seems to have been a rise in the number of cases. I am aware of a couple of workarounds. Tell me, does this adapter have bluetooth capability or is it wifi only?

Side note: You won't like this but the only sure way to eliminate multi-state (storage mode) as an issue is to get an adapter that is single-state. The Main Menu for this site is as follows:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 can help. Menu item 2 gives information and links to many adapters that are single-state.

With that said, answer my question and I will try to help.

Massedil, 2024-12-16:
> I'll try to help.

Thank you !

> Tell me, does this adapter have bluetooth capability or is it wifi only?

It is Wi-Fi only. Id with `lsusb` is `0bda:c811`. ["AC600 driver free" is written on it](https://duckduckgo.com/?t=ffab&q=AC600+driver+free&iax=images&ia=images). 

> You won't like this but the only sure way to eliminate multi-state (storage mode) as an issue is to get an adapter that is single-state. 

But can't this bug be solved in driver, kernel or somewhere else ?
And why the bug doesn't occur if my PC is already booted when I plug the device ?

> With that said, answer my question and I will try to help.

Done, let me know if I can do more for help solving this ! 

Thanks again !

morrownr, 2024-12-16:
> But can't this bug be solved in driver, kernel or somewhere else ?

The bug cannot be solved in the driver as the driver has nothing to do with what causes the issue. In fact, you will see the same problem without the driver installed. Let me post a short guide that I have in the FAQ of one of the drivers here:

-----

Question: My USB WiFi adapter is showing up as a CDROM or Flash drive instead of a WiFi adapter. What is the problem?

Answer: Your Realtek USB WiFi adapter showing up as a CDROM or Flash drive (often with ID 0bda:1a2b) instead of functioning as a network adapter (such as ID 35bc:0102 or similar) is likely due to a "mode-switching" issue. Some USB WiFi adapters include onboard memory that contains drivers or installation software for Windows. When plugged into a system for the first time, they initially present themselves as a virtual CD-ROM or Flash driver containing the drivers.

In Linux, the usb_modeswitch utility generally handles this issue but there are situations where it does not work as expected.

Your options:

1. Send your adapter back and get one that is single-state (no storage onboard). If that is not possible then:

2. You need to check if the VID/PID for your adapter is in the usb_modeswitch data file. See the following link for more information:

https://github.com/morrownr/USB-WiFi/blob/main/home/How_to_Modeswitch.md

If you have exhausted recommendations from the information in item 2, then:

3. You may be able to make the adapter work in wifi mode with the following method:

How to add kernel parameters with GRUB in Ubuntu:

Note: This method may vary by distro so if you are not using Ubuntu, please consult the documentation for your distro.

Once your device has booted, use a text editor to open the following file:

$ sudo nano /etc/default/grub

Add parameters to GRUB_CMDLINE_LINUX while keeping the following in mind:

Enter parameters inside the double-quotes

Leave a space before each new parameter

Don’t add space round = and other punctuations for each key-value

Don’t add line breaks

If your original line looks like:

GRUB_CMDLINE_LINUX="quiet"

Then your updated line should read like:

GRUB_CMDLINE_LINUX="quiet usb-storage.quirks=0bda:1a2b:i"

Note: If the storage VId/PID (ID) i not 0bda:1a2b, you will need to change 0bda:1a2b to the storage mode VID/PID of your adapter.

Save and close the editor.

Update GRUB with its new configuration:

$ sudo update-grub

$ sudo reboot

Note: If your distro does not use grub, the RasPiOS is an example, you will need to read your distro docs to see how to do the above.

Example for Raspberry Pi OS:

$ sudo nano /boot/firmware/cmdline.txt

add the following to the end of the console= line:

usb-storage.quirks=0bda:1a2b:i

Save, close the editor and reboot.

-----

Massedil, 2024-12-16:
Thanks !

```
$ grep 0bda /usr/lib/udev/rules.d/40-usb_modeswitch.rules 
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="usb_modeswitch '/%k'"
```

And nothing with `c811` in the file.

That is why I see `1a2b` in `déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00` ?

So you suggest me to try : `GRUB_CMDLINE_LINUX="usb-storage.quirks=0bda:c811:i"` ?

Things I don't understand :
- why my problem is random and the device is not always seen as a CD-ROM ?
- why the problem doesn't occurs when the PC is already started when I plug the device ?

Massedil, 2024-12-17:
Thanks !

```
$ grep 0bda /usr/lib/udev/rules.d/40-usb_modeswitch.rules 
ATTR{idVendor}=="0bda", ATTR{idProduct}=="1a2b", RUN+="usb_modeswitch '/%k'"
```

And nothing with `c811` in the file.

So you suggest me to try : `GRUB_CMDLINE_LINUX="usb-storage.quirks=0bda:c811:i"` ?

Things I don't understand :
- why my problem is random and the device is not always seen as a CD-ROM ?
- why the problem doesn't occurs when the PC is already started when I plug the device ?

morrownr, 2024-12-17:
> And nothing with c811 in the file.

There should be nothing in that file with c811. usb_modeswitch is try to switch you out of CDROM mode so it needs the ID of the storage.

> - why my problem is random and the device is not always seen as a CD-ROM ?

It is not unusual for this problem to be random. 

> - why the problem doesn't occurs when the PC is already started when I plug the device ?

That does give us a hint that something could be getting messed up during the boot process. usb_modeswitch is started by systemd.

> I've done GRUB_CMDLINE_LINUX="usb-storage.quirks=0bda:c811:i", rebooted, and for this reboot, it works fine !

That actually should not work as your should be using the ID for the storage mode but if it works, great. How about you use it some more and see if it was coincident.

> I've also activate EnableLogging=1 in /etc/usb_modeswitch.conf, but I can't find the logs... Do you know how to see them ?

I think the log is somewhere down in /var and you should be able to see it with `sudo nano <file>`.

Massedil, 2024-12-17:
> usb_modeswitch is started by systemd

I don't see anything in logs about "usb_modeswitch" when I boot. I think usb_modeswitch is not started at all, that is why I don't have logs.

But when I plug the device after boot, tadam : 

```
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: Created slice system-usb_modeswitch.slice - Slice /system/usb_modeswitch.
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: Starting usb_modeswitch@1-5.service - USB_ModeSwitch_1-5...
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: usb_modeswitch@1-5.service: Main process exited, code=killed, status=15/TERM
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: usb_modeswitch@1-5.service: Failed with result 'signal'.
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: Stopped usb_modeswitch@1-5.service - USB_ModeSwitch_1-5.
déc. 17 01:33:51 XXXXXXXXXX systemd[1]: Starting usb_modeswitch@1-5.service - USB_ModeSwitch_1-5...
déc. 17 01:33:52 XXXXXXXXXX usb_modeswitch[4839]: switch device 0bda:1a2b on 001/007
```

Now I have a log file at `/var/log/usb_modeswitch_1-5` : 

```
USB_ModeSwitch log from Tue Dec 17 01:33:51 CET 2024
Raw parameters: --switch-mode 1-5
Use global config file: /etc/usb_modeswitch.conf

Use global config file: /etc/usb_modeswitch.conf
Use top device dir /sys/bus/usb/devices/1-5
Check class of first interface ...
 Interface class is 08.

----------------
USB values from sysfs:
  manufacturer	Realtek
  product	DISK
  serial	
----------------
bNumConfigurations is 1 - don't check for active configuration
Found packed config collection /usr/share/usb_modeswitch/configPack.tar.gz
ConfigList: pack/0bda:1a2b pack/
SCSI attributes not needed, move on
Check config: pack/0bda:1a2b
! matched. Read config data
Extract config 0bda:1a2b from collection /usr/share/usb_modeswitch/configPack.tar.gz
Command line:
usb_modeswitch -W -D -u -1 -b 1 -g 7 -v 0bda -p 1a2b -f $flags(config)

Verbose debug output of usb_modeswitch and libusb follows
(Note that some USB errors are to be expected in the process)
--------------------------------

Read long config from command line

 * usb_modeswitch: handle USB devices with multiple modes
 * Version 2.6.1 (C) Josua Dietze 2017
 * Based on libusb1/libusbx

 ! PLEASE REPORT NEW CONFIGURATIONS !

DefaultVendor=  0x0bda
DefaultProduct= 0x1a2b
TargetVendor=   0x2001
TargetProduct=  0x331d

StandardEject=1
System integration mode enabled

Use given bus/device number: 001/007 ...
Look for default devices ...
 bus/device number matched
  found USB ID 0bda:1a2b
   vendor ID matched
   product ID matched
 Found devices in default mode (1)
Get the current device configuration ...
Use interface number 0
 with class 8
Use endpoints 0x0b (out) and 0x8a (in)

USB description data (for identification)
-------------------------
Manufacturer: Realtek
     Product: DISK
  Serial No.: not provided
-------------------------
Sending standard EJECT sequence
Looking for active drivers ...
 OK, driver detached
Set up interface 0
Use endpoint 0x0b for message sending ...
Trying to send message 1 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 1 (CSW) ...
 Response successfully read (13 bytes), status 1
Trying to send message 2 to endpoint 0x0b ...
 OK, message successfully sent
Read the response to message 2 (CSW) ...
 Response successfully read (13 bytes), status 0
Trying to send message 3 to endpoint 0x0b ...
 Sending the message returned error -1. Try to continue
Read the response to message 3 (CSW) ...
 Device seems to have vanished after reading. Good.
 Device is gone, skip any further commands
ok:busdev
--------------------------------
(end of usb_modeswitch output)

Check success of mode switch for max. 20 seconds ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...
 Read attributes ...

Target config not matching - current values are
    1-5:1.0/bInterfaceClass:   ff
    bConfigurationValue:       1
    bNumConfigurations:        1
    busnum:                    1
    devnum:                    8
    idProduct:                 c811
    idVendor:                  0bda
    manufacturer:              Realtek
    product:                   802.11ac NIC
    serial:                    123456

Mode switching may have failed. Exit
```

> That actually should not work as your should be using the ID for the storage mode but if it works, great. How about you use it some more and see if it was coincident.

Yes, surely a coincidence. A good boot as describe in my first post.

Now, I need to find why it is not started at boot...

```
$ locate usb_modeswitch
/etc/usb_modeswitch.conf
/etc/usb_modeswitch.d
/usr/lib/systemd/system/usb_modeswitch@.service
/usr/lib/udev/usb_modeswitch
/usr/lib/udev/rules.d/40-usb_modeswitch.rules
/usr/sbin/usb_modeswitch
/usr/sbin/usb_modeswitch_dispatcher
/usr/share/usb_modeswitch
/usr/share/man/man1/usb_modeswitch.1.gz
/usr/share/man/man1/usb_modeswitch_dispatcher.1.gz
/usr/share/usb_modeswitch/configPack.tar.gz
/var/lib/usb_modeswitch
```

Do you have an idea ?

morrownr, 2024-12-17:
> Now, I need to find why it is not started at boot...  Do you have an idea ?

I do not have a good idea. I have been digging some as I have had time over the last week but obviously I have not uncovered the problem. I'm digging for another chip, the rtl8852/32cu. I just went public with a new driver for that chip today and several users have reported usb_modeswitch problems and I am seeing it myself. Same stuff that you and others see with this chip. If you want to work together to figure out what is causing this, I'm in.

One thing we need to do is document things as we go along. I end up working on other things so I need a document to refresh my mind.

So what starts usb_modeswitch? systemd. So, let's look at the service file usbmodeswitch:

/usr/lib/systemd/system/usb_modeswitch@.service

Now editing the original may not be a good idea so copying this original to a place that will override the original will give us a file to edit and if we mess it up we can just copy the original over the one we are editing and start over.

sudo cp /usr/lib/systemd/system/usb_modeswitch@.service /etc/systemd/system/usb_modeswitch@.service

Then to edit:

sudo nano /etc/systemd/system/usb_modeswitch@.service

So, what do we edit? Beats me. We need to figure out where things are going wrong.

morrownr, 2024-12-17:
Here is an interesting read. Not saying this is our problem but it adds to our knowledge:

https://ubuntuforums.org/archive/index.php/t-2481847.html

morrownr, 2024-12-17:
Another interesting read:

https://superuser.com/questions/1210501/how-can-i-start-usb-modeswitch-before-the-network-configuration

Massedil, 2024-12-17:
Just to inform you that I didn't have time today (and just quickly read your messages), but I will try to understand the boot process and debug the problem. I'll let you informed of what I will find. Thanks again !

morrownr, 2024-12-17:
No hurry. I had a little extra time and found a few things that might add something to the search.

Massedil, 2024-12-18:
Hello,

Some news !

I better understand my cases.

When I cold boot (real power off), I have the problem.

The device present it self as a CD-ROM `1a2b`. usb_modswitch don't execute at all :

```
kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
```

When I reboot, I don't have the problem because I imagine the usb device is not power off and retain the last usb_modswitch switch command and presents immediately itself as `c811` : 

```
déc. 18 22:04:17 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
```

usb_modswitch don't execute at all in this case !

***So in all cases, usb_modswitch is not executed at boot.***

When I have booted and plug the device (here with udev logs activated), everything works fine because udev trigger a usb_modeswitch call :

```
déc. 18 22:08:41 XXXXXXXXXX systemd-udevd[379]: Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
déc. 18 22:08:43 XXXXXXXXXX systemd-udevd[379]: Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
déc. 18 22:08:48 XXXXXXXXXX systemd-udevd[379]: Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
déc. 18 22:08:51 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: /usr/lib/udev/rules.d/40-usb_modeswitch.rules:417 RUN 'usb_modeswitch '/%k''
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Running command "usb_modeswitch '/1-5'"
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Starting 'usb_modeswitch '/1-5''
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: Created slice system-usb_modeswitch.slice - Slice /system/usb_modeswitch.
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: Starting usb_modeswitch@1-5.service - USB_ModeSwitch_1-5...
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Process 'usb_modeswitch '/1-5'' succeeded.
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: /usr/lib/udev/rules.d/40-usb_modeswitch.rules:417 RUN 'usb_modeswitch '/%k''
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Running command "usb_modeswitch '/1-5'"
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Starting 'usb_modeswitch '/1-5''
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: usb_modeswitch@1-5.service: Main process exited, code=killed, status=15/TERM
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: usb_modeswitch@1-5.service: Failed with result 'signal'.
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: Stopped usb_modeswitch@1-5.service - USB_ModeSwitch_1-5.
déc. 18 22:08:51 XXXXXXXXXX (udev-worker)[1516]: 1-5: Process 'usb_modeswitch '/1-5'' succeeded.
déc. 18 22:08:51 XXXXXXXXXX systemd[1]: Starting usb_modeswitch@1-5.service - USB_ModeSwitch_1-5...
déc. 18 22:08:52 XXXXXXXXXX usb_modeswitch[1637]: switch device 0bda:1a2b on 001/005
déc. 18 22:08:52 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
déc. 18 22:09:12 XXXXXXXXXX systemd[1]: usb_modeswitch@1-5.service: Deactivated successfully.
déc. 18 22:09:12 XXXXXXXXXX systemd[1]: Finished usb_modeswitch@1-5.service - USB_ModeSwitch_1-5.
```

Massedil, 2024-12-18:
Some things to understand or find where to report if it is a bug : 
- [ ] Why usb_modswitch is not called at boot ?
- [ ] Why the kernel is failing at boot with the CD-ROM in a way that block connexion to the user session ? (I can't use my keyboard and mouse !)

Massedil, 2024-12-18:
I found that udev is well called at boot. But it seems there is some deadlock when trying to apply actions on the CD-ROM `sr0` and the USB port `1-5` : 

```
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: new high-speed USB device number 2 using xhci_hcd
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: New USB device strings: Mfr=1, Product=2, SerialNumber=0
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: Product: DISK
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: Manufacturer: Realtek
déc. 18 22:07:44 XXXXXXXXXX kernel: usb-storage 1-5:1.0: USB Mass Storage device detected
déc. 18 22:07:44 XXXXXXXXXX kernel: scsi host8: usb-storage 1-5:1.0
déc. 18 22:07:44 XXXXXXXXXX kernel: sr 8:0:0:0: [sr0] scsi3-mmc drive: 0x/0x caddy
déc. 18 22:07:44 XXXXXXXXXX kernel: sr 8:0:0:0: Attached scsi CD-ROM sr0
déc. 18 22:07:44 XXXXXXXXXX udevadm[334]: 1-5: Triggered device with action 'add'.
déc. 18 22:07:44 XXXXXXXXXX udevadm[334]: 1-5:1.0: Triggered device with action 'add'.
déc. 18 22:07:44 XXXXXXXXXX udevadm[334]: sr0: Triggered device with action 'add'.
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: Device is queued (SEQNUM=3274, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3274 blocked by SEQNUM=3266
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: Device is queued (SEQNUM=3278, ACTION=change)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3278 blocked by SEQNUM=3266
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3274 blocked by SEQNUM=3267
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3278 blocked by SEQNUM=3267
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: Device ready for processing (SEQNUM=3274, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX (udev-worker)[549]: sr0: Processing device (SEQNUM=3274, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: sd-device-monitor(manager): Passed 280 byte to netlink monitor.
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3278 blocked by SEQNUM=3274
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: Device is queued (SEQNUM=3428, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: SEQNUM=3428 blocked by SEQNUM=3274
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5:1.0: Device is queued (SEQNUM=3429, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5:1.0: SEQNUM=3429 blocked by SEQNUM=3274
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: Device is queued (SEQNUM=3433, ACTION=add)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: sr0: SEQNUM=3433 blocked by SEQNUM=3274
déc. 18 22:07:44 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: Device is queued (SEQNUM=4780, ACTION=change)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: SEQNUM=4780 blocked by SEQNUM=3274
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: Device is queued (SEQNUM=4805, ACTION=change)
déc. 18 22:07:44 XXXXXXXXXX systemd-udevd[388]: 1-5: SEQNUM=4805 blocked by SEQNUM=3274
```

I reboot and try to wait more to see what happens. In this boot sequence I halted the system at 22:08:02.

Massedil, 2024-12-18:
> Here is an interesting read. Not saying this is our problem but it adds to our knowledge:
> https://ubuntuforums.org/archive/index.php/t-2481847.html

Common behavior, when an old boot failed (without udev logs) : 

```
déc. 16 15:38:27 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:38:58 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:39:28 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:39:59 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:40:30 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:40:40 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:41:11 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:41:41 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:42:12 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:42:43 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:43:14 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:43:44 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:44:15 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
déc. 16 15:44:22 XXXXXXXXXX kernel: usb 1-5: reset high-speed USB device number 2 using xhci_hcd
```

Given the long time between messages, that is why I will wait more for next fail reboot !

---
> Another interesting read:
>
> https://superuser.com/questions/1210501/how-can-i-start-usb-modeswitch-before-the-network-configuration

May be a hack we can try if we didn't find the problem.

---

I have hope that we will have more info after next failed reboot with udev debug logs !

Massedil, 2024-12-18:
Waiting a long time finally permit me to log-in !

In `lsusb`, I have :

```
Bus 001 Device 002: ID 0bda:1a2b Realtek Semiconductor Corp. RTL8188GU 802.11n WLAN Adapter (Driver CDROM Mode)
```

In fact, we can see that usb_modeswitch is called :

```
déc. 18 23:30:35 PaRaZiTe (udev-worker)[1452]: 1-5: Running command "usb_modeswitch '/1-5'"
déc. 18 23:31:12 PaRaZiTe (udev-worker)[1598]: 1-5: Running command "usb_modeswitch '/1-5'"
[more than 30 times the same line]
déc. 18 23:31:12 PaRaZiTe (udev-worker)[1598]: 1-5: Running command "usb_modeswitch '/1-5'"
```

I have the log of the first call only (don't understand why not the last) :

```
-rw-r--r-- 1 root root 830 déc.  18 23:30 /var/log/usb_modeswitch_1-5
```

Content of `/var/log/usb_modeswitch_1-5` :

```
USB_ModeSwitch log from Wed Dec 18 23:30:35 CET 2024
Raw parameters: --switch-mode 1-5
Use global config file: /etc/usb_modeswitch.conf

Use global config file: /etc/usb_modeswitch.conf
Use top device dir /sys/bus/usb/devices/1-5
Check class of first interface ...
 Interface class is 08.

----------------
USB values from sysfs:
  manufacturer	Realtek
  product	DISK
  serial	
----------------
bNumConfigurations is 1 - don't check for active configuration
Found packed config collection /usr/share/usb_modeswitch/configPack.tar.gz
ConfigList: pack/0bda:1a2b pack/
SCSI attributes not needed, move on
Check config: pack/0bda:1a2b
! matched. Read config data
Extract config 0bda:1a2b from collection /usr/share/usb_modeswitch/configPack.tar.gz
Command line:
usb_modeswitch -W -D -u -1 -b 1 -g 2 -v 0bda -p 1a2b -f $flags(config)
```

So we can see that the command is not really launched compared to the log I have when the session has booted.

Massedil, 2024-12-18:
Here are the log file.

I hope it gives you ideas.

[udevlog.boot.fail_wait_long.log](https://github.com/user-attachments/files/18190530/udevlog.boot.fail_wait_long.log)

morrownr, 2024-12-19:
@Massedil 

Looking at the log without knowing what you were doing with the computer makes it hard to interpret but...

I think it shows that there is difficulty bringing SR0 up. That is, the system is having a problem bringing what it thinks is a CDROM up. This in turn causes usb_modeswitch from being triggered and that means no mode switching is taking place.

This is interesting. On 3 test systems here, the adapter I am testing works fine on a system that has a real CDROM but fails on the other two that do not. Coincidence? I don't know but can I get you to clean the udev log and only do one cold boot. Send the log from just doing that.

I'll try to do the same on two systems here as able. I am having some health difficulties so I may be slow over the next two days or so..

Massedil, 2024-12-19:
> Looking at the log without knowing what you were doing with the computer makes it hard to interpret but...

Simple : in udevlog.boot.fail_wait_long.log I have done nothing ! Just booted and waited doing nothing. In this file you have a cold boot with the device plugged in.

> I think it shows that there is difficulty bringing SR0 up. That is, the system is having a problem bringing what it thinks is a CDROM up.

I agree. 

> This in turn causes usb_modeswitch from being triggered and that means no mode switching is taking place.

For me usb_modeswitch is triggered (but fail I don't know why), because we can see :

```
déc. 18 23:30:35 XXXXXXXXXX (udev-worker)[1452]: 1-5: Running command "usb_modeswitch '/1-5'"
```
Or:
```
déc. 18 23:30:35 XXXXXXXXXX systemd[1]: Starting usb_modeswitch@1-5.service - USB_ModeSwitch_1-5...
déc. 18 23:30:35 XXXXXXXXXX (udev-worker)[1452]: 1-5: Process 'usb_modeswitch '/1-5'' succeeded.
```

> This is interesting. On 3 test systems here, the adapter I am testing works fine on a system that has a real CDROM but fails on the other two that do not. Coincidence? 

I effectively don't have a real CD-ROM plugged in, but may be I can plug one if you need.

> I don't know but can I get you to clean the udev log and only do one cold boot. Send the log from just doing that.

You have that in the file. Only one cold boot from déc. 18 23:21:56 to déc. 18 23:31:12.

Let me know if I can help more.

morrownr, 2024-12-21:
I am going to see about getting some help. It may take some time. We need to get to the bottom of this.

morrownr, 2025-01-02:
I hope you had a good holiday season. I am no further along than I was we left off. I am just holding this issue until I discover more information. I think it is a problem with Linux bringing up the cd-rom. If the cd-rom is not up like it should be then usb_modeswitch can't switch it.

Massedil, 2025-01-10:
Hi, 

> I hope you had a good holiday season.

Yes, thanks, I hope you too !

Some news.

I changed by mistake the USB port (after moving my computer) now it is 1-8, before it was 1-5. 

I works... !

May be because 1-5 is USB-3 socket (not sure, because I don't remember what exact socket is was) and 1-8 is USB-2 socket (sure !) ?

(Also, kernel is updated to `6.8.0-51-generic`. But I don't think it is related)

For me, it looks like a race condition, but why it is not triggered anymore now... ?

Here are the log.

First detection :

```
janv. 10 16:16:43 XXXXXXXXXX kernel: usb 1-8: new high-speed USB device number 4 using xhci_hcd
janv. 10 16:16:43 XXXXXXXXXX kernel: usb 1-8: New USB device found, idVendor=0bda, idProduct=1a2b, bcdDevice= 2.00
janv. 10 16:16:43 XXXXXXXXXX kernel: usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=0
janv. 10 16:16:43 XXXXXXXXXX kernel: usb 1-8: Product: DISK
janv. 10 16:16:43 XXXXXXXXXX kernel: usb 1-8: Manufacturer: Realtek
janv. 10 16:16:43 XXXXXXXXXX kernel: usb-storage 1-8:1.0: USB Mass Storage device detected
janv. 10 16:16:43 XXXXXXXXXX kernel: scsi host8: usb-storage 1-8:1.0
```

```
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: Parsed configuration file "/usr/lib/systemd/network/73-usb-net-by-mac.link"
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: Reading rules file: /usr/lib/udev/rules.d/39-usbmuxd.rules
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: Reading rules file: /usr/lib/udev/rules.d/40-usb_modeswitch.rules
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: Reading rules file: /usr/lib/udev/rules.d/60-cdrom_id.rules
janv. 10 16:16:43 XXXXXXXXXX udevadm[331]: USBC000:00: Triggered device with action 'add'.
janv. 10 16:16:43 XXXXXXXXXX udevadm[331]: 1-8: Triggered device with action 'add'.
janv. 10 16:16:43 XXXXXXXXXX udevadm[331]: 1-8:1.0: Triggered device with action 'add'.
janv. 10 16:16:43 XXXXXXXXXX udevadm[331]: usb2: Triggered device with action 'add'.
```

```
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: sr: Device is queued (SEQNUM=3830, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: sr: SEQNUM=3830 blocked by SEQNUM=3828
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: sr: Device ready for processing (SEQNUM=3830, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: sr: sd-device-monitor(manager): Passed 172 byte to netlink monitor.
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[392]: sr: Processing device (SEQNUM=3830, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[392]: sr: Device processed (SEQNUM=3830, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[392]: sr: sd-device-monitor(worker): Passed 172 byte to netlink monitor.
```

First : `RUN 'usb_modeswitch '/%k'`

```
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: 1-8: Device is queued (SEQNUM=4289, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[533]: hiddev0: Preserve permissions of /dev/usb/hiddev0, uid=0, gid=0, mode=0600
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[533]: hiddev0: Successfully created symlink '/dev/char/180:0' to '/dev/usb/hiddev0'
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[391]: hidraw2: sd-device: Created db file '/run/udev/data/c241:2' for '/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.2/0003:0458:017D.0003/hidraw/hidraw2'
janv. 10 16:16:43 XXXXXXXXXX systemd-udevd[373]: 1-8: Device ready for processing (SEQNUM=4289, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[533]: hiddev0: sd-device: Created empty file '/run/udev/data/c180:0' for '/devices/pci0000:00/0000:00:14.0/usb1/1-6/1-6:1.1/usbmisc/hiddev0'
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[390]: 1-8: Processing device (SEQNUM=4289, ACTION=add)
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[390]: 1-8: Removing watch handle -1.
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[393]: hiddev1: Preserve permissions of /dev/usb/hiddev1, uid=0, gid=0, mode=0600
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[393]: hiddev1: Successfully created symlink '/dev/char/180:1' to '/dev/usb/hiddev1'
janv. 10 16:16:43 XXXXXXXXXX (udev-worker)[390]: 1-8: /usr/lib/udev/rules.d/40-usb_modeswitch.rules:417 RUN 'usb_modeswitch '/%k''
```

CD-ROM is detected by kernel :
[... 319 lines cut ...]
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: sd-device: Created db file '/run/udev/data/c189:3' for '/devices/pci0000:00/0000:00:14.0/usb1/1-8'
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Device processed (SEQNUM=5151, ACTION=unbind)
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: sd-device-monitor(worker): Passed 434 byte to netlink monitor.
janv. 10 16:17:24 XXXXXXXXXX systemd-udevd[373]: 1-8: Device ready for processing (SEQNUM=5152, ACTION=remove)
janv. 10 16:17:24 XXXXXXXXXX systemd-udevd[373]: 1-8: sd-device-monitor(manager): Passed 300 byte to netlink monitor.
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Processing device (SEQNUM=5152, ACTION=remove)
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Removing watch handle -1.
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: /usr/lib/udev/rules.d/60-virtualbox.rules:4 RUN 'VBoxCreateUSBNode.sh --remove $major $minor'
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Running command "VBoxCreateUSBNode.sh --remove 189 3"
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Starting 'VBoxCreateUSBNode.sh --remove 189 3'
janv. 10 16:17:24 XXXXXXXXXX (udev-worker)[2499]: 1-8: Process 'VBoxCreateUSBNode.sh --remove 189 3' succeeded.
[... cut ...]
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: new high-speed USB device number 6 using xhci_hcd
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: New USB device found, idVendor=0bda, idProduct=c811, bcdDevice= 2.00
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: New USB device strings: Mfr=1, Product=2, SerialNumber=3
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: Product: 802.11ac NIC
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: Manufacturer: Realtek
janv. 10 16:17:24 XXXXXXXXXX kernel: usb 1-8: SerialNumber: 123456
```

morrownr, 2025-01-11:
> For me, it looks like a race condition, but why it is not triggered anymore now... ?

Another mystery. If I see a problem with an adapter, I usually change ports. USB2 can be more stable. 

One thing I recommend to USB WiFi adapter users is avoid multi-state adapters if possible. I have seen too many problems with adapters that have storage mode.

Here is the Main Menu for my site:

https://github.com/morrownr/USB-WiFi

Reading menu items 1 and 2 could be useful.

Massedil, 2025-01-11:
> Here is the Main Menu for my site:
>
> https://github.com/morrownr/USB-WiFi
>
> Reading menu items 1 and 2 could be useful.

Thanks, I had already read them. But I'm in the case where I bought a device "Linux compatible" without knowing in advance what will be the exact model.

> One thing I recommend to USB WiFi adapter users is avoid multi-state adapters if possible.

I understand you recommend to avoid them, but the best would be a full Linux support ! 

Don't you see we have enough log (and ways to reproduce the problem) to open an issue in [kernel bugzilla](https://bugzilla.kernel.org/) ?

morrownr, 2025-01-11:
> Don't you see we have enough log (and ways to reproduce the problem) to open an issue in kernel bugzilla ?

I'd say the chance of that fixing the problem is about zero. Which kernel dev is going to work this problem? None. Why? Because this is not a driver problem per se. 

I did not wake up one day last month and decide to recommend that Linux users avoid multi-state (windows driver onboard) adapters. I've been using Linux since 1994. Multi-state adapters can be problematic for Windows users and Linux users. Some adapters are more problematic than others. You happen to have one of the most problematic in that it has a rtl8821/11cu chip.

When you buy a product, your dollars are votes to tell the maker that we like what they are making. Don't vote for products that don't work well or you will get more of the same. Staying far away from multi-state adapters is the solution that can be most effective and is within our power to do. I have seen evidence that seems to support that this strategy is working as several adapter makers are making at least some models that meet the criteria set forth at this site:

- avoid multi-state adapters

- avoid multi-function adapters (no bluetooth)

- buy adapters that have good quality in-kernel drivers so that they are plug and play.

The Plug and Play List shows multiple adapter makers that have adapters that meet this criteria... even Netgear and TP-Link has select adapters that meet this criteria. Try not to fight the issue, think about what I am saying.

Massedil, 2025-01-27:
I have maybe found something interesting [here](https://forums.linuxmint.com/viewtopic.php?p=2565492&sid=b4957039cd0a3c2434b74133c50fa680#p2565492) : 

> Boot delays
> 
> In rare circumstances, it’s possible that the Linux boot or shutdown process may delay for a long time while trying to access that “flash storage”. To check if that’s the case, remove the adapter from the USB slot and see if the delay is gone.
> 
> Normally, installing our driver and using an up to date distribution should avoid this issue, but if it happens, the first thing to try is to run this command and reboot:
> 
> sudo update-initramfs -u
> 
> If that didn’t resolve the issue, then the second thing to try is to add the following parameter in grub, which instructs the kernel to completely ignore the flash storage mode of our adapters:
> 
> usb-storage.quirks=0bda:1a2b:i,a69c:5721:i

So may be we can solve (hide ?) the problem with `usb-storage.quirks=0bda:1a2b:i` kernel parameter.