morrownr/8821cu-20210916 #175: Incompatible with linux 6.14-rc3

Salz, 2025-02-17:
Due to commit [7a53af85d3bbd](https://git.kernel.org/pub/scm/linux/kernel/git/torvalds/linux.git/commit/?h=v6.14-rc3&id=7a53af85d3bbd) in the linux kernel, the driver no longer build with the following error:

```
  CC [M]  os_dep/linux/ioctl_cfg80211.o
os_dep/linux/ioctl_cfg80211.c:10527:25: error: initialization of ‘int (*)(struct wiphy *, struct wireless_dev *, unsigned int,  int *)’ from incompatible pointer type ‘int (*)(struct wiphy *, struct wireless_dev *, int *)’ [-Wincompatible-pointer-types]
10527 |         .get_tx_power = cfg80211_rtw_get_txpower,
      |                         ^~~~~~~~~~~~~~~~~~~~~~~~
os_dep/linux/ioctl_cfg80211.c:10527:25: note: (near initialization for ‘rtw_cfg80211_ops.get_tx_power’)
make[4]: *** [/usr/src/linux-headers-6.14.0-rc3-spatz20250216/scripts/Makefile.build:207: os_dep/linux/ioctl_cfg80211.o] Error 1
```

The following patch fixes the build on 6.14-rc3:

```
diff -Ndurp ./os_dep/linux/ioctl_cfg80211.c /usr/src/rtl8821cu-5.12.0.4/./os_de>
--- ./os_dep/linux/ioctl_cfg80211.c     2025-02-04 08:36:58.206644347 +0100
+++ /usr/src/rtl8821cu-5.12.0.4/./os_dep/linux/ioctl_cfg80211.c 2025-02-17 13:5>
@@ -4650,6 +4650,9 @@ static int cfg80211_rtw_get_txpower(stru
 #if (LINUX_VERSION_CODE >= KERNEL_VERSION(3, 8, 0))
        struct wireless_dev *wdev,
 #endif
+#if (LINUX_VERSION_CODE >= KERNEL_VERSION(6, 14, 0))
+       unsigned int link_id,
+#endif
        int *dbm)
 {
        struct dvobj_priv *dvobj = wiphy_to_dvobj(wiphy);
```

I didn't get around to test if the driver still works there, but it does on 6.13.0, and so does finally the official rtw88_8821cu. But thanks for your work on this hardware.

morrownr, 2025-02-18:
Hi @Salz 

Thanks for the info. I need to test on 6.14 rc3. I tested on an earlier version of 6.14 and did not see this so I need to narrow down when this came about.

> I didn't get around to test if the driver still works there, but it does on 6.13.0, and so does finally the official rtw88_8821cu.

FYI: I am looking at discontinuing API updates on this driver and the other Realtek WiFi 5 out-of-kernel drivers that I maintain here after 6.13 as rtw88 in 6.14 is at the point where all of the WiFi 5 drivers in rtw88 are better than the drivers here.

Salz, 2025-02-19:
I had the same error on 6.14-rc1 but decided to wait a few -rc releases for the changed api to probably be final and I didn't need the driver in that machine, since it has on board wifi.

I build and install my kernel with make bindeb-pkg, install the resulting linux-image-* and linux-headers-* packages, and use dkms for the 8821cu module, all on devuan unstable. Maybe there are differences in my build setup so I caught it but not you…

morrownr, 2025-02-19:
> I had the same error on 6.14-rc1...

Let me retest. I do so many similar things that sometimes I mix things up. I just want to make sure this patch goes in with the correct info.

morrownr, 2025-02-19:
@Salz 

I tested and you are correct. This problem started with the beginning of 6.14. I have added the patch and will test it as I have time today. Thanks.