morrownr/8821cu-20210916 #183: How to make two access points (2.4GHz + 5GHz) on openwrt?

SaintAnd, 2025-07-05:
Hello friends!!!
I have a USB wi-fi adapter **tenda ac650 u10**. I was able to configure using standard **rtw88-8821cu** drivers only **one** access point either 2.4 GHz or 5 GHz. Is it possible to create **two access points** at once on this adapter?

There is only one USB port here.  

Mikrotik **RB760iGS** that does not have built-in Wi-Fi

ImmortalWRT 21.10.2 - Used

SaintAnd, 2025-07-05:
> Hi [@SaintAnd](https://github.com/SaintAnd)
Thank you very much for your answer! I thought maybe there is a way to make virtual devices and periodically switch the radio point to frequencies, but apparently it will not be practical. Probably easier to install a USB hub and connect another USB-Wi-Fi :)))

P.S. Thanks for the drivers. On some versions of OpenWRT the code does not compile due to kernel version errors, but with the help of chatagpt everything is fixed, thanks!

closed 2025-07-05 by SaintAnd

morrownr, 2025-07-05:
https://www.raspberrypi.com/products/usb-3-hub/

SaintAnd, 2025-07-06:
> @morrownr  Also, The Plug and Play List on the Main Menu could be of help as the Mediatek chips are well supported in OpenWRT.

Thanks for the tip!!! Unfortunately, they don't deliver directly to Russia, Belarus and Ukraine, so I can try to order through Aliexpress.
By the way, the standard rtl8xxxu drivers don't work for my 2.4 GHz TP-link WN722N v2 modem, so I'll have to compile them manually. Thanks again

```
[ 2566.511859] usb 1-1: This Realtek USB WiFi dongle (0x2357:0x010c) is untested!
[ 2566.519259] usb 1-1: Please report results to Jes.Sorensen@gmail.com
[ 2566.660035] usb 1-1: Fatal - failed to parse EFuse
[ 2566.664885] rtl8xxxu: probe of 1-1:1.0 failed with error -22
```