morrownr/8821cu-20210916 #184: rtl8821cu-morrownr-dkms-git error during newest 6.16 kernel installation

johnpiers, 2025-08-15:
I see that the last time the maintainer at the ARCH AUR mikl-ls last logged in was: 2024-09-21 (UTC) that's almost a year ago. Is this package out-of-date? (This is not really relevant to GitHub I'm sue but thought I'd mention it)

As per *** below with the current (6.16) linux-zen kernel and current linux-zen-headers I get this

==> WARNING: `dkms install --no-depmod rtl8821cu/5.12.0.4.203.945c687 -k 6.16.0-zen2-1-zen' exited 10

If anyone can advise that'll be great. Thanks.

*** I get this error when upgrading to the newest Linux Kernel 6.16 both 'linux' and 'linux-zen'.

==> dkms install --no-depmod rtl8821cu/5.12.0.4.203.945c687 -k 6.16.0-zen2-1-zen

Deprecated feature: CLEAN (/usr/src/rtl8821cu-5.12.0.4.203.945c687/dkms.conf)

Error! Bad return status for module build on kernel: 6.16.0-zen2-1-zen (x86_64) Consult /var/lib/dkms/rtl8821cu/5.12.0.4.203.945c687/build/make.log for more information.

==> WARNING: `dkms install --no-depmod rtl8821cu/5.12.0.4.203.945c687 -k 6.16.0-zen2-1-zen' exited 10

(The installed dkms module is: rtl8821cu-morrownr-dkms-git 5.12.0.4.203.945c687-1)

I've rolled the kernel (zen) and kernel (zen) headers back to 6.15.9

morrownr, 2025-08-15:
Hi @johnpiers 

I am curious why you are installing an out-of-kernel driver on a kernel that contains a nice new modern in-kernel driver for your hardware. Were you aware of this?

johnpiers, 2025-08-15:
> Hi [@johnpiers](https://github.com/johnpiers)
> 
> I am curious why you are installing an out-of-kernel driver on a kernel that contains a nice new modern in-kernel driver for your hardware. Were you aware of this?

I did not [very embarrassing]. So I can just uninstall it  and be done with it?

morrownr, 2025-08-15:
> So I can just uninstall it and be done with it?

That is the theory. Make sure you uninstall any previous drivers you have installed. I cannot help with drivers from ARCH AUR as it is not using my installer... which has, over the years, really made me mad because I have gone to trouble to make extensive docs and then.... well, enough of that.

If that does not work, let me know and we will see what is the cause.

closed 2025-08-15 by johnpiers

johnpiers, 2025-08-15:
Ok, so then do you [[morrownr](https://github.com/morrownr)] think that this should be reopened [pending] ?  Mainly because of bcosca comment a few minutes ago.
Thanks again.

johnpiers, 2025-08-15:
All good then https://github.com/morrownr. Appreciate your replies. Have a good one.
Best Regards,
John.